raspberrypi/pico-examples #510: Issue in flash_program.c example

lautisilber, 2024-08-08:
The functions ```flash_range_erase``` and ```flash_range_program``` are run without being surrounded by a 
```
uint32_t interrupts = save_and_disable_interrupts();

...

restore_interrupts(interrupts);
```
block and in my tests, the program crashes without these.

(I'm sorry. This is my first time rising an issue. If this behaviour is excepted or my testing is wrong, sorry for the inconvenience)

peterharperuk, 2024-09-03:
You are correct that if the processor is likely to run any code from flash when calling these functions, then they will cause a crash. However this program isn't doing anything else and so doesn't crash. Did you really reproduce a crash running this example?! Anyway - I will improve the example with some comments so anyone copying the code is aware of the problem.

lautisilber, 2024-09-03:
The demo run just fine. Only when I tried to copy paste it into a larger project it crashed without those lines. Sorry for the inconvenience and thank you for listening :)

peterharperuk, 2024-09-03:
I think you're right that we should fix the example to show the proper way to do things. So I've reopened the issue and I'll update the example

kilograham, 2024-11-22:
merged into develop

PR #545 (2024-11-22): Use flash_safe_execute
Although it's not required in this example, demonstrate how to call flash function using flash_safe_execute

Fixes #510
commit b6303d83 (2024-11-22): Use flash_safe_execute (#545)

* Use flash_safe_execute

Although it's not required in this example, demonstrate how to call
flash function using flash_safe_execute

Fixes #510
closed 2024-11-22 by kilograham