raspberrypi/pico-examples #513: Building for PICO_BOARD=pico2 fails on Windows

ndabas, 2024-08-10:
I've tried with my own DIY toolchain and the official VS Code extension. Building pico-examples on Windows fails at the end of the CMake configure process:

```
[cmake] -- Configuring done (4.5s)
[cmake] -- Generating done (3.2s)
[cmake] CMake Error:
[cmake]   Running
[cmake] 
[cmake]    'C:/Users/WDAGUtilityAccount/.pico-sdk/ninja/v1.12.1/ninja.exe' '-C' 'C:/Users/WDAGUtilityAccount/Downloads/pico-examples/build' '-t' 'recompact'
[cmake] 
[cmake]   failed with:
[cmake] 
[cmake]    ninja: warning: phony target 'bootloaders/encrypted/enc_bootloader.ld' names itself as an input; ignoring [-w phonycycle=warn]
[cmake] 
[cmake]   ninja: warning: phony target 'bootloaders/encrypted/hello_serial_enc.ld'
[cmake]   names itself as an input; ignoring [-w phonycycle=warn]
[cmake] 
[cmake]   ninja: error: build.ninja:11266: multiple rules generate
[cmake]   bootloaders/encrypted/enc_bootloader.ld
[cmake] 
[cmake]   
[cmake] 
[cmake] 
[cmake] 
[cmake] CMake Generate step failed.  Build files cannot be regenerated correctly.
```

I tried with CMake 3.28 and also 3.30, same result.

matsobdev, 2024-08-12:
Same here (CMake 3.19.8), but for RISC-V Embedded (riscv32-unknown-elf) on Windows 7 SP1 x64 finishes mostly ok:
```
In function 'ip_reass_dequeue_datagram',
    inlined from 'ip4_reass' at E:/Download/Pico_2/pico-sdk-2.0.0/lib/lwip/src/core/ipv4/ip4_frag.c:663:5:
E:/Download/Pico_2/pico-sdk-2.0.0/lib/lwip/src/core/ipv4/ip4_frag.c:326:16: warning: potential null pointer dereference [-Wnull-dereference]
  326 |     prev->next = ipr->next;
      |     ~~~~~~~~~~~^~~~~~~~~~~
In function 'ip_reass_dequeue_datagram',
    inlined from 'ip4_reass' at E:/Download/Pico_2/pico-sdk-2.0.0/lib/lwip/src/core/ipv4/ip4_frag.c:684:5:
E:/Download/Pico_2/pico-sdk-2.0.0/lib/lwip/src/core/ipv4/ip4_frag.c:326:16: warning: null pointer dereference [-Wnull-dereference]
  326 |     prev->next = ipr->next;
      |     ~~~~~~~~~~~^~~~~~~~~~~
```
but "it's only a glitch".

ndabas, 2024-08-12:
@matsobdev that looks like a completely separate issue. Would be good to create a new issue, because like I clarified, the problem I see is not specific to Windows.

lurch, 2024-08-12:
> Edit: I first tried this on Windows but this is also failing on Linux platforms. So not a Windows-specific issue. It's the same error on Linux too.

Looks like it might be specific to Ninja?

On Ubuntu 24.04 if I run cmake with:
```
cmake .. -DPICO_SDK_PATH=../../pico-sdk -DPICO_BOARD=pico2
```
then `cmake` completes successfully, and running `make` (eventually) completes successfully too.
But if I create a separate build-directory and invoke cmake with:
```
cmake .. -DPICO_SDK_PATH=../../pico-sdk -DPICO_BOARD=pico2 -GNinja
```
then I get _exactly_ the same error as you do above about `multiple rules generate bootloaders/encrypted/enc_bootloader.ld` :confused:  I'm running cmake 3.22.1 and ninja 1.10.1

Ahhh.... but if I create yet another build directory and run:
```
cmake .. -DPICO_SDK_PATH=../../pico-sdk -DPICO_BOARD=pico -GNinja
```
then this time `cmake` completes successfully, and running `ninja` builds all the examples. So looks like there's something specific about the linker scripts for RP2350 that Ninja doesn't like, but Make is perfectly happy with?

ndabas, 2024-08-13:
Yes, true:
- Generating Makefiles and building with `make` works fine for all targets;
- Building my own projects with `PICO_BOARD=pico2` works fine with both `make` and `ninja`;
- Building pico-examples with Ninja and `PICO_BOARD=pico2` fails.

The `build.ninja` contains these two definitions to build the target in question:

```
#############################################
# Utility command for enc_bootloader.ld

build bootloaders/encrypted/enc_bootloader.ld: phony bootloaders/encrypted/CMakeFiles/enc_bootloader.ld bootloaders/encrypted/enc_bootloader.ld
```

```
#############################################
# Custom command for bootloaders\encrypted\enc_bootloader.ld

build bootloaders/encrypted/enc_bootloader.ld | ${cmake_ninja_workdir}bootloaders/encrypted/enc_bootloader.ld: CUSTOM_COMMAND C$:/Program$ Files/Raspberry$ Pi/Pico$ SDK$ v2.0.0/pico-sdk/src/rp2_common/pico_crt0/rp2350/memmap_no_flash.ld
  COMMAND = C:\Windows\system32\cmd.exe /C "cd /D C:\Users\WDAGUtilityAccount\Documents\Pico-v2.0.0\pico-examples\build-pico2\bootloaders\encrypted && "C:\Program Files\Raspberry Pi\Pico SDK v2.0.0\cmake\bin\cmake.exe" "-DPICO_LINKER_SCRIPT_PATH:PATH=C:/Program Files/Raspberry Pi/Pico SDK v2.0.0/pico-sdk/src/rp2_common/pico_crt0/rp2350" -Dorigin="0x20070000" -Dlength="64k" -Doutput_file:FILEPATH=C:/Users/WDAGUtilityAccount/Documents/Pico-v2.0.0/pico-examples/build-pico2/bootloaders/encrypted/enc_bootloader.ld -P C:/Users/WDAGUtilityAccount/Documents/Pico-v2.0.0/pico-examples/build-pico2/bootloaders/encrypted/make_linker_script.cmake"
  DESC = Generating enc_bootloader.ld
  restat = 1
```

I guess these are generated by these lines in the source:
https://github.com/raspberrypi/pico-examples/blob/7fe60d6b4027771e45d97f207532c41b1d8c5418/bootloaders/encrypted/CMakeLists.txt#L33
https://github.com/raspberrypi/pico-examples/blob/7fe60d6b4027771e45d97f207532c41b1d8c5418/bootloaders/encrypted/CMakeLists.txt#L26

I'm not really sure if this is a bug in CMake's Ninja generator, but it is a problem for building these examples on Windows, if I don't want to use `nmake` from Visual Studio or similar. Ninja is just a lot faster + easier to install.

will-v-pi, 2024-08-16:
This should be fixed with that commit - literally just needed `.` changed to `_` in the target name…

kilograham, 2024-08-17:
fixed in develop

closed 2024-08-17 by kilograham

ndabas, 2024-08-17:
@will-v-pi @kilograham great, thanks!

commit 4b39afbb (2024-08-16): Fix #513 - ninja compilation of enc_bootloader

Ninja was treating ${target}.ld and ${CMAKE_CURRENT_BINARY_DIR}/${target}.ld as the same file, despite the first just being a target name not a file - fixed by changing . to _