raspberrypi/pico-examples #516: Genera; usage documentation issues

znmeb, 2024-08-16:
I just got a SparkFun Pro RP2350 and I'm trying to get the examples to build. I've run into a few issues I could fix via web search, but there are some outstanding ones:

1. RISC-V compilers: The README says to use stable but the C/C++ SDK manual PDF says to use top-of-tree. Who's right? Will both of them work?
2. The CMake step says: `Skipping universal examples as PICO_RISCV_TOOLCHAIN_PATH and PICO_ARM_TOOLCHAIN_PATH are not defined`. Where is the documentation on how to set these?
3. CMake: `Skipping TinyUSB dual examples, as TinyUSB hw/mcu/raspberry_pi/Pico-PIO-USB submodule unavailable`. How do I get that submodule?
4. FreeRtOS: 
    
```
Using FREERTOS_KERNEL_PATH from environment ('/var/home/znmeb/dbx-homes/CLAMS-PicoSDK/pico/FreeRTOS-Kernel')
CMake Error at freertos/FreeRTOS_Kernel_import.cmake:87 (message):
  Directory '/var/home/znmeb/dbx-homes/CLAMS-PicoSDK/pico/FreeRTOS-Kernel'
  does not contain a 'rp2350-arm-s' port here:
  portable/ThirdParty/Community-Supported-Ports/GCC/RP2350_ARM_NTZ
```

How do I get the rp2350-arm-s port? A web search didn't turn anything up, so I assume either it hasn't been released yet or the search engines haven't indexed it.

lurch, 2024-08-16:
pinging @nathan-contino for questions about the documentation and @will-v-pi for questions about the RISC-V compilers.

> CMake: Skipping TinyUSB dual examples, as TinyUSB hw/mcu/raspberry_pi/Pico-PIO-USB submodule unavailable. How do I get that submodule?

Hmmm, TinyUSB _used_ to use submodules, but it looks like it now uses a dedicated script for downloading dependencies (see https://docs.tinyusb.org/en/latest/reference/getting_started.html#dependencies ) so I guess the message in https://github.com/raspberrypi/pico-examples/blob/master/usb/CMakeLists.txt needs updating.

So it seems the complete set of steps is:
```console
git clone https://github.com/raspberrypi/pico-sdk.git
cd pico-sdk
git submodule update --init
cd lib/tinyusb
python3 tools/get_deps.py rp2040
```

will-v-pi, 2024-08-16:
1. My personal recommendation for Risc-V compilers (and what the VS Code extension uses) would be the Core-V top of tree [here](https://www.embecosm.com/resources/tool-chain-downloads/#corev), or for Pi my pico-sdk-tools one from [here](https://github.com/raspberrypi/pico-sdk-tools/releases/tag/v2.0.0-0)
2. It’s not documented - just set them as CMake variables (`-DPICO_ARM_TOOLCHAIN_PATH=…`) - they’re not part of the SDK, just part of the Universal Binary examples, as those require both an Arm and a RISC-V compiler
3. Pinging @liamfraser as he was looking into this?
4. Use [this kernel](https://github.com/raspberrypi/FreeRTOS-Kernel) for now, there are plans to get the port merged upstream, but I don’t think it’s happened yet

znmeb, 2024-08-16:
> 1. My personal recommendation for Risc-V compilers (and what the VS Code extension uses) would be the Core-V top of tree [here](https://www.embecosm.com/resources/tool-chain-downloads/#corev), or [...]

I'm using the top-of-tree because I built my setup scripts using the SDK manual and didn't want to use stable, which is older.

> 
>     2. It’s not documented - just set them as CMake variables (`-DPICO_ARM_TOOLCHAIN_PATH=…`) - they’re not part of the SDK, just part of the Universal Binary examples, as those require both an Arm and a RISC-V compiler

Ah ... that's what I figured. But which PATH? `/usr/local` and it finds everything or `/usr/local/bin` or something else.
> 
>     3. Pinging @liamfraser as he was looking into this?
> 
>     4. Use [this kernel](https://github.com/raspberrypi/FreeRTOS-Kernel) for now, there are plans to get the port merged upstream, but I don’t think it’s happened yet

Yeah - I found that repo after I opened the issue and I have it working now.

will-v-pi, 2024-08-16:
> Ah ... that's what I figured. But which PATH? `/usr/local` and it finds everything or `/usr/local/bin` or something else?

The same path you’d use for `PICO_TOOLCHAIN_PATH`, which is documented in the C SDK book. If your compiler is `/usr/local/bin/arm-none-eabi-gcc` then `/usr/local` should work

znmeb, 2024-08-17:
> > Ah ... that's what I figured. But which PATH? `/usr/local` and it finds everything or `/usr/local/bin` or something else?
> 
> The same path you’d use for `PICO_TOOLCHAIN_PATH`, which is documented in the C SDK book. If your compiler is `/usr/local/bin/arm-none-eabi-gcc` then `/usr/local` should work

I couldn't get this to work. I looked at the source and it looked to me like there was only one example in the "universal" set, an example of how to build a `uf2` that would run either as Arm or RISC-V. That's not something I have an immediate need for, so I can wait for a newer release.

I also looked at the TinyUSB dual case. It doesn't look like there are any examples yet, just examples for host and device. So again, I can wait for a newer release. You can close this unless you want me to do more troubleshooting on the universal case, since I have hardware.

lurch, 2024-08-18:
> I also looked at the TinyUSB dual case. It doesn't look like there are any examples yet, just examples for host and device.

This part of [README.md](https://github.com/raspberrypi/pico-examples/blob/develop/README.md#usb-dual-mode) says otherwise?
[image]

Although perhaps that also needs updating, as https://github.com/hathach/tinyusb/tree/master/examples/dual now has `host_hid_to_device_cdc` and `host_info_to_device_cdc`.

nathan-contino, 2024-11-07:
Do we need to update any of the PDF or HTML doc to clarify this, or has it all been addressed elsewhere?