raspberrypi/pico-examples #520: system/rand/rand.c doesn't cross-compile on x64 Ubuntu

jdbruner, 2024-08-18:
I'm unable to compile `system/rand/rand.c` on Ubuntu 24.04, x86_64, because the build uses the newlib version of `inttypes.h`, which does not define `PRIx64` (The compiler unhelpfully suggests including `inttypes.h`, because the native version in `/usr/include` does define this.)

kilograham, 2024-08-18:
what version of arm-none-eabi-gcc are you using; it should have PRIx64 in inttypes.h

jdbruner, 2024-08-18:
```
$ apt show gcc-arm-none-eabi
Package: gcc-arm-none-eabi
Version: 15:13.2.rel1-2
Priority: extra
Section: universe/devel
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Agustin Henze <tin@debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 517 MB
Depends: libc6 (>= 2.38), libgcc-s1 (>= 3.3.1), libgmp10 (>= 2:6.3.0+dfsg), libisl23 (>= 0.15), libmpc3 (>= 1.1.0), libmpfr6 (>= 3.1.3), libstdc++6 (>= 5), zlib1g (>= 1:1.1.4), binutils-arm-none-eabi
Recommends: libnewlib-arm-none-eabi
Breaks: libnewlib-dev (<= 2.4.0.20160527-4), libstdc++-arm-none-eabi-newlib (<= 15:6.3.1+svn253039-1+10)
Homepage: https://developer.arm.com/open-source/gnu-toolchain/gnu-rm
Download-Size: 57.5 MB
APT-Manual-Installed: yes
APT-Sources: http://us.archive.ubuntu.com/ubuntu noble/universe amd64 Packages
Description: GCC cross compiler for ARM Cortex-R/M processors
 Bare metal C and C++ compiler for embedded ARM chips using Cortex-M, and
 Cortex-R processors.
 This package is based on the GNU ARM toolchain provided by ARM.

$ apt show libnewlib-dev
Package: libnewlib-dev
Version: 4.4.0.20231231-2
Priority: extra
Section: universe/libdevel
Source: newlib
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian QA Group <packages@qa.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 959 kB
Suggests: libnewlib-arm-none-eabi, gcc-arm-none-eabi
Breaks: libnewlib0
Replaces: libnewlib0
Homepage: https://sourceware.org/newlib/
Download-Size: 140 kB
APT-Manual-Installed: no
APT-Sources: http://us.archive.ubuntu.com/ubuntu noble/universe amd64 Packages
Description: C library and math library intended for use on embedded systems
 Newlib is a C library and math library intended for use on embedded systems.
 It is a conglomeration of several library parts, all under free software
 licenses that make them easily usable on embedded products.
 .
 This package contains the newlib headers.
```

lurch, 2024-08-19:
@kilograham I just tested this in an Ubuntu 24.04 Docker image, and I was able to reproduce this by trying to build the `develop` branch of pico-sdk. `gcc-arm-none-eabi` and `libnewlib-dev` versions are identical to those shown above.
Commands I ran (inside the docker container) were:
```console
apt update
apt install build-essential git cmake ninja-build gcc-arm-none-eabi python3
git clone --depth 1 -b develop https://github.com/raspberrypi/pico-sdk.git
git clone --depth 1 -b develop https://github.com/raspberrypi/pico-examples.git
cd pico-sdk
mkdir build
cd build
cmake .. -DPICO_EXAMPLES_PATH=../../pico-examples
make
```
and the build failed with:
```
[  1%] Built target bs2_default_padded_checksummed_asm
[  1%] Building C object test/pico_stdlib_test/CMakeFiles/pico_stdlib_test.dir/pico_stdlib_test.c.o
/pico-sdk/test/pico_stdlib_test/pico_stdlib_test.c: In function 'main':
/pico-sdk/test/pico_stdlib_test/pico_stdlib_test.c:92:20: error: expected ')' before 'PRIu64'
   92 |         printf( "%" PRIu64 "\n", to_us_since_boot(get_absolute_time()));
      |               ~    ^~~~~~~
      |                    )
/pico-sdk/test/pico_stdlib_test/pico_stdlib_test.c:12:1: note: 'PRIu64' is defined in header '<inttypes.h>'; did you forget to '#include <inttypes.h>'?
   11 | #include <stdlib.h>
  +++ |+#include <inttypes.h>
   12 | 
/pico-sdk/test/pico_stdlib_test/pico_stdlib_test.c:98:19: error: expected ')' before 'PRIu64'
   98 |         printf("%" PRIu64 "\n", to_us_since_boot(get_absolute_time()));
      |               ~   ^~~~~~~
      |                   )
/pico-sdk/test/pico_stdlib_test/pico_stdlib_test.c:98:20: note: 'PRIu64' is defined in header '<inttypes.h>'; did you forget to '#include <inttypes.h>'?
   98 |         printf("%" PRIu64 "\n", to_us_since_boot(get_absolute_time()));
      |                    ^~~~~~
make[2]: *** [test/pico_stdlib_test/CMakeFiles/pico_stdlib_test.dir/build.make:76: test/pico_stdlib_test/CMakeFiles/pico_stdlib_test.dir/pico_stdlib_test.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2104: test/pico_stdlib_test/CMakeFiles/pico_stdlib_test.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

If I perform the _exact_ same steps inside an Ubuntu 22.04 Docker image (where `gcc-arm-none-eabi` is `Version: 15:10.3-2021.07-4` and `libnewlib-dev` is `Version: 3.3.0-1.3`) then the build of `pico-sdk` runs to completion.

kilograham, 2024-08-19:
seems like a compiler/newlib version bug - it has happened in the past

I have fixed with a workaround

closed 2024-08-19 by kilograham