raspberrypi/pico-examples #528: Output OTP decryption key JSON file for `encrypted` example always uses example key (`00010203...`)

0xTJ, 2024-08-23:
In the example at `bootloaders/encrypted`, the README.md file gives instructions on how to generate a new AES key with the command:
```
dd if=/dev/urandom of=privateaes.bin bs=1 count=32
```
However, the `otp.json` file in the CMake build binary directory, which contains the data to be programmed into the decryption key OTP rows , is simply a copy of the `otp.json` in the example directory.
This can be confusing to the user since, if (the CMake binary directory copy of) `otp.json` is used to program OTP, the example key will be permanently burned into OTP, when they might expect the new key to be used instead.

This is particularly confusing as the signing key is correctly filed out (by picotool) to `hello_serial_enc.otp.json`.
I believe that there should either be a note indicating that the key in `otp.json` will not be updated with a newly-generated key, or some script/instructions for generating a decryption key OTP file (from `privateaes.bin`) should be added.

Octopus1633, 2024-09-29:
Hello,

I defined a new key and wrote the new key to an unused page in otp. The otp.json file was also updated. I rewrote the otp.json file to otp and it showed that the write failed, but the encryption program with the new key can run normally (the encryption program cannot be run when it is not written), which seems to have written the new key to the corresponding page of otp.
[image]

In addition, I tested writing the otp.json file in the build directory to otp and performed a secure boot. I found that the test results are the same as non-secure boot. Unable to run the unencrypted program in flash, but can run in SRAM. Is this normal?

will-v-pi, 2024-10-15:
Merged into develop

PR #536 (2024-10-01): Add cmake script to update otp.json from privateaes.bin
Currently, any changes to the key in privateaes.bin will not be reflected in the otp.json file, which could lead to issues if that is written to OTP.

This adds a cmake script using the JSON string handling to update the otp.json file if privateaes.bin is modified, before copying it to the binary directory. This fixes #528 

This will not work on cmake versions prior to 3.19, as those don't support JSON string handling, so instead it throws a warning if the file has been changed. This should only throw the warning if privateaes.json has b [... cut]
closed 2024-10-15 by will-v-pi