raspberrypi/pico-examples #530: Missing an example to set low power states on Pico 2

ullibak, 2024-08-25:
I try to enable the low power states on the Pico 2 but without any success. If I read the dokumentation correctly, power state P1.7 equals 1111 in binary or 15 decimal. Using 

```
uint32_t ps = 15;
int i = powman_set_power_state(ps);
```
returns 0 for i which means that the function was performed successfully, but adding

`printf("Power State: %d\n", powman_get_power_state ());`

right after the code above returns 0, so that somehow, the power state has not changed.

I am sure that I am missing something so it would be nice to add an example that demonstrates how the Pico 2 could be sent to a low power state for a definded amount of time.

Thanks!

ullibak, 2024-08-26:
The sleep demo you mention uses some "tricks" to save power an I can not see how the new power management states of the RP2350 are used. Reading the specs of the RP2350 I and the SDK documentation, I was expecting some much more high level approach to send the RP2350 into an low power state. So a simple example in the SDK would be very helpful.

peterharperuk, 2024-08-27:
Those sleep examples don't use powman. I don't think we're going to add a powman example. We want to add a new API for low power but ran out of time before the Pico 2 sdk update. In the meantime there are a couple powman examples here https://github.com/peterharperuk/pico-examples/commit/35db8fc31d322de2bcac641d9b6b0f6fdb73062c

ullibak, 2024-08-28:
Hi
Thanks a lot! The powerman examples are really useful and helped a lot to undestand how to use the new power states.
With the timer example, I am able to lower the power consumption to 0.7 mA during sleep. This is good, but still a multiple of the values given in section 4.1.1 of the data sheet: 

https://datasheets.raspberrypi.com/pico/pico-2-datasheet.pdf

Power state P1.7 should draw only 0.148 mA.

In the data sheet they describe the current consumption measurements with

_All externally accessible GPIOs, SWD and SWCLK are unconnected. The RP2350’s USB PHY has been powered down,
and the DP and DM pull-downs enabled prior to entering the low power state. The USB cable remains connected to a
host computer._

I have no GPIO connected. But I do not know how to power down the USB PHY and how to enable the DP and DM pull-downs.

Does anyone know how to do this?

Thanks!

lurch, 2024-08-28:
> But I do not know how to power down the USB PHY

That's probably related to section 6.1.4. in https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf ? However note that it says "This supply must always be provided, even in applications where the USB PHY is never used."

ullibak, 2024-08-28:
Thanks, lurch
Section 6.1. describes the different power supplies needed to run the RP2350 chip. But from the text in section 4.1.1. of the Pico 2 data sheet I conclude that there must be a way to power down USB PHY using a software command.

peterharperuk, 2024-08-28:
I will try and find out where those numbers come from. The best I can achieve is .29mA, by powering the device with 3V3 and disconnecting everything (USB, UART, SWD) and removing R10 (at your own risk)

ullibak, 2024-08-28:
I can get 0.36 mA when connecting a Li-Ion battery (4.2V) to VSYS and leave everything else disconnected. 

In section 1.2.2. of the RP2350 data sheet all the pins are described. The description for the USP_DM and USB_DP pins is:

_USB controller, supporting Full Speed device and Full/Low Speed host. A 27Ω series termination
resistor is required on each pin, but bus pullups and pulldowns are provided internally. These pins
can be used as software-controlled GPIOs, if USB is not required._

So there is a way to connect internal pulldowns via software commands. But I could not figure out how to to this.

peterharperuk, 2024-09-04:
I can reproduce the power numbers in the datasheet with this commit. Make sure you disconnect everything except for 5v into vsys https://github.com/peterharperuk/pico-examples/commit/7dccd00d15ded4ddf961f44fdcd1f11a9d8c8be1

closed 2024-09-04 by peterharperuk

tannewt, 2024-09-04:
> I can reproduce the power numbers in the datasheet with this commit. Make sure you disconnect everything except for 5v into vsys [peterharperuk@7dccd00](https://github.com/peterharperuk/pico-examples/commit/7dccd00d15ded4ddf961f44fdcd1f11a9d8c8be1)

Nice! How are others expected to reproduce this? Copy that code? Use a SDK API?

ullibak, 2024-09-05:
Thanks a lot!
Li-Ion battery (4.11 V) connected to VSYS. Pico draws 150.8 microamps during low power. This is 620 microwatts. Great!!!

@tannewt: You find the code in https://github.com/peterharperuk/pico-examples/tree/35db8fc31d322de2bcac641d9b6b0f6fdb73062c/powman

it is emedded in the example directory tree there. You can copy th whole directory tree and substitute pico-examples.

I use a simplified standalone version, see attached zip file. 
[powman_standalone.zip](https://github.com/user-attachments/files/16885755/powman_standalone.zip)

copy these two files plus pico_sdk_import.cmake and pico_extras_import_optional.cmake from the Pico SDK into a directory and you should be able to compile as usual. I use VS Code with the Pico Extension.

peterharperuk, 2024-09-05:
There might be a non-hardware API in future. So you can wait for that or else you'll have to copy that code.

matsobdev, 2024-11-30:
Powering flash down will allow to save extra 10 uA (mileage may vary), but then when starting from flash, it should be powered back again: https://github.com/raspberrypi/pico-sdk/issues/2111

peterharperuk, 2026-03-02:
150uA. That's what this closed issue is discussing.
What "Discussions" are you referring to?