raspberrypi/pico-examples #532: how to setup the SysTick handler

oldoldman, 2024-08-27:
Hi guys,
I'm trying to setup SysTick handler on a picow board. What I konw roughly is
1. Setup the SysTick by programming the syst_csr/syst_rvr registers
2. Setup the IRQ handler by calling irq_set_enabled
When it comes to the irq_set_enabled, I have not idea of which irq number shoud I use. Any suggestion? Many thanks!

lurch, 2024-08-27:
See [here](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_hardware_irq_1interrupt_nums) for a full list of interrupt numbers. (Or page 188 in https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf )

oldoldman, 2024-08-27:
I've found from [here](https://github.com/Blimp01/pico_non_blocking_timer/blob/master/non_blocking_timer.c#L16) that if I "override" the isr_systick in [crt0.s](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_crt0/crt0.S#L47), the SysTick will work. So I guess the irq number of SysTick is SIO_IRQ_PROC0. Is there any doc about this?

kilograham, 2024-08-27:
SysTick is not an interrupt, it is an exception

see here https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_exception/include/hardware/exception.h

oldoldman, 2024-08-28:
But this is from the ARMv6-M arch doc
[image]
So SysTick is an interrupt. But it is not that important.

kilograham, 2024-08-28:
Ha; from the same document ;-)

[image]

but as you say, it doesn't matter, however from an API perspective, the SDK treats it like an exception rather than an _external_ interrupt

oldoldman, 2024-08-28:
I my understanding of the ARM doc, interrupt is a sub-category of exception. So they are used interchangably.

kilograham, 2024-08-28:
yeah, it is confusing, because it lives with the synchronous exceptions but acts like an interrupt, but doesn't have the regular NVIC controls like the rest of the interrupts. Anyways, my point was that the `hardware_irq` deals with the external interrupts (and has 0 based indexes for those)