raspberrypi/pico-examples #544: RP2350 does not respond to flash_nuke.uf2

TRadigk, 2024-09-03:
Experiment: Copy over the flash_nuke.uf2 to RP2350 based PICO 2 board 
Observation: nothing happens. 
Expected behaviour: Controller is rebooted and flash is empty. 
Actual behaviour: Nothing happens, the file explorer remains open, until any other uf2 is copied which contains executable code.

lurch, 2024-09-03:
Seems to work okay for me. Are you sure that you're using the latest https://datasheets.raspberrypi.com/soft/flash_nuke.uf2 (which is now a "[universal binary](https://github.com/raspberrypi/pico-examples?tab=readme-ov-file#universal)") and not the older version (which was RP2040-only) ? Old version had a size of 25600 bytes, new version has a size of 98304 bytes.

peterharperuk, 2024-09-03:
If you built it for RP2040 it would behave exactly as descrived

TRadigk, 2024-09-03:
You are right, this page is not updated: https://www.elektronik-kompendium.de/sites/raspberry-pi/2611111.htm
(second google result and first where I learned that this file exists in the first place)

lurch, 2024-09-03:
You'll be pleased to hear that https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html definitely *is* kept up to date! :joy:

TRadigk, 2024-09-03:
You got me a new bookmark 😁 👍