raspberrypi/pico-examples #549: `fatal error: pico/cyw43_arch.h: No such file or directory` when compiling blink for Pico W

SengerM, 2024-09-18:
I am following instructions from [here](https://www.raspberrypi.com/documentation/microcontrollers/c_sdk.html). I am working on a Raspberry Pi 4 with this OS version:

```
$ cat /etc/rpi-issue 
Raspberry Pi reference 2024-07-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 48efb5fc5485fafdc9de8ad481eb5c09e1182656, stage4
```

I want to compile `blink` for Pico W. First I did:

```
myself@pi4u:~/code/pico-examples/build $ cmake .. -DPICO_SDK_PATH=../../pico-sdk -DPICO_BOARD=pico_w
PICO_SDK_PATH is /home/myself/code/pico-sdk
Target board (PICO_BOARD) is 'pico_w'.
Using board configuration from /home/myself/code/pico-sdk/src/boards/include/boards/pico_w.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
Build type is Release
CMake Warning at /home/myself/code/pico-sdk/src/rp2_common/tinyusb/CMakeLists.txt:10 (message):
  TinyUSB submodule has not been initialized; USB support will be unavailable

  hint: try 'git submodule update --init' from your SDK directory
  (/home/myself/code/pico-sdk).

CMake Warning at /home/myself/code/pico-sdk/src/rp2_common/pico_btstack/CMakeLists.txt:10 (message):
  btstack submodule has not been initialized; Pico W BLE support will be
  unavailable.

              hint: try 'git submodule update --init' from your SDK directory (/home/myself/code/pico-sdk).

CMake Warning at /home/myself/code/pico-sdk/src/rp2_common/pico_cyw43_driver/CMakeLists.txt:11 (message):
  cyw43-driver submodule has not been initialized; Pico W wireless support
  will be unavailable

  hint: try 'git submodule update --init' from your SDK directory
  (/home/myself/code/pico-sdk).

CMake Warning at /home/myself/code/pico-sdk/src/rp2_common/pico_lwip/CMakeLists.txt:10 (message):
  LWIP submodule has not been initialized; Pico W wireless support will be
  unavailable

  #hint: try 'git submodule update --init' from your SDK directory
  (/home/myself/code/pico-sdk).

Skipping hello_usb because TinyUSB submodule is not initialized in the SDK
Skipping encrypted bootloader example as pico_mbedtls unavailable
Skipping hello_dcp example which is unsupported on this platform
Skipping runtime_flash_permissions example which is unsupported on this platform
Skipping dvi_out_hstx_encoder example which is unsupported on this platform
Skipping spi_lcd example which is unsupported on this platform
Skipping multicore_doorbell example which is unsupported on this platform
Skipping hello_otp example which is unsupported on this platform
Skipping Pico W examples as support is not available
Skipping boot_info example which is unsupported on this platform
Skipping universal examples as PICO_RISCV_TOOLCHAIN_PATH and PICO_ARM_TOOLCHAIN_PATH are not defined
Skipping TinyUSB device examples as TinyUSB is unavailable
Skipping TinyUSB host examples as TinyUSB is unavailable
Skipping TinyUSB dual examples, as TinyUSB hw/mcu/raspberry_pi/Pico-PIO-USB submodule unavailable
Skipping SHA256 examples as pico_sha256 or pico_mbedtls unavailable
Skipping FreeRTOS examples as FREERTOS_KERNEL_PATH not defined
-- Configuring done
-- Generating done
-- Build files have been written to: /home/myself/code/pico-examples/build
```

and then

```
myself@pi4u:~/code/pico-examples/build $ make blink
Built target bs2_default
Built target bs2_default_padded_checksummed_asm
Building C object blink/CMakeFiles/blink.dir/blink.c.obj
/home/myself/code/pico-examples/blink/blink.c:12:10: fatal error: pico/cyw43_arch.h: No such file or directory
   12 | #include "pico/cyw43_arch.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [blink/CMakeFiles/blink.dir/build.make:76: blink/CMakeFiles/blink.dir/blink.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:4103: blink/CMakeFiles/blink.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:4110: blink/CMakeFiles/blink.dir/rule] Error 2
make: *** [Makefile:163: blink] Error 2
```

How can I fix this?

lurch, 2024-09-18:
Did you try doing the `hint: try 'git submodule update --init' from your SDK directory (/home/myself/code/pico-sdk).` that it suggests?

SengerM, 2024-09-18:
I missed that hint, thanks for pointing it out. After doing that it is now working.

closed 2024-09-18 by SengerM