raspberrypi/pico-examples #550: pio quadrature_encoder example  not working on pico 2 RP2350

danielallstar, 2024-09-18:
The pio - quadrature_encoder example does not seem to work with the raspberry pi pico 2 (RP2350). It does however work with the pico RP2040.

The expected behaviour is that when you turn a quarature encoder the "new_value" changes. This does however not happen with the pico 2

Any help to resolve this issue

lurch, 2024-09-18:
ping @pmarques-dev who originally wrote this example.

pmarques-dev, 2024-09-19:
I haven't got a RP2350 yet, but have one ordered and shipping at the moment. When that arrives, I can test it and figure out what's going on

peterharperuk, 2024-09-19:
I've reproduced the problem - checking what's going on.

pmarques-dev, 2024-09-19:
from the top of my head, I would verify that the code is actually being successfully loaded at address 0. Maybe the new SDK uses PIO's for something and the example is silently failing to load the code and then trying to use it anyway

pmarques-dev, 2024-09-20:
if I remember correctly, to use the pins as inputs you don't have to multiplex them to the PIO, at least on the RP2040, so maybe there is a difference in behaviour there

peterharperuk, 2024-09-20:
Yes - this is covered in the datasheet, section 9.3, it's a change from rp2040. Calling gpio_init also fixes the problem. I will submit a patch.

> The pad reset state is different from RP2040, which only disables digital inputs on GPIOs 26 through 29 (as of
version B2) and does not have isolation latches. Applications must enable the pad input (GPIO0.IE = 1) and disable
pad isolation latches (GPIO0.ISO = 0) before using the pads for digital I/O. The gpio_set_function() SDK function
performs these tasks automatically.

pmarques-dev, 2024-09-20:
this in fact is in the SDK documentation, so it may need some clarification there as well:
https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_hardware_pio_1gafa244b1be8f53a329db9d26298e054bb
"Note that this is not necessary for a state machine to be able to read the input value from a GPIO, but only for it to set the output value or output enable."

peterharperuk, 2024-09-20:
Well arguably that's correct, you just need to have setup the gpio for input.

danielallstar, 2024-09-21:
Thank you very much for looking into the problem and quickly finding a solution! 

I did a test and now the pio quadrature_encoder example works as expected on RP2350.

PR #551 (2024-09-20): Fix quadrature_encoder examples on rp2350
Have to initialise the gpio function for rp2350.
Stop the quadrature_encoder example spamming sdio if nothing has changed.
Change the pins used for quadrature_encoder_substep so they match the other example.

Fixes #550
commit 362f6766 (2024-09-20): Fix quadrature_encoder examples on rp2350 (#551)

Have to initialise the gpio function for rp2350.
Stop the quadrature_encoder example spamming sdio if nothing has
changed.
Change the pins used for quadrature_encoder_substep so they match the
other example.

Fixes #550
closed 2024-09-21 by danielallstar