raspberrypi/pico-examples #552: bootloaders/encrypted routines are not functioning properly

Octopus1633, 2024-09-25:
Hello, 

When I run the bootloaders/encrypted routine, the encrypted program is not run at the end. Is this normal? What is the problem?
[image]

will-v-pi, 2024-09-25:
Did you write the generated otp.json file (in the build folder with the UF2s) to your OTP with `picotool otp load otp.json`? This example requires the AES key to be stored in OTP, else the decryption will not work - the generated otp.json file will write this key, lock the relevant OTP page, and enable secure boot on your device. I'll add this to the readme for this example

Note that writing the One-Time-Programable memory this way will only allow binaries signed with the private key used to run on your device.

lurch, 2024-09-26:
So this issue can be closed?

Octopus1633, 2024-09-26:
of course.

closed 2024-09-26 by Octopus1633