raspberrypi/pico-examples #554: otp/hello_otp example problem

Octopus1633, 2024-09-26:
Hi, I am running the otp/hello_otp example, why does the output show that the first write fails for both ECC and RAW, is this normal, is there any problem?
[image]

will-v-pi, 2024-09-26:
Is this your second time running the example? This example will only work correctly when run for the first time on a board as it modifies the OTP (One-Time-Programable Memory), so if you run it a second time on the same board it will throw these errors - this bootrom error is documented in the SDK

https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/pico_bootrom/include/pico/bootrom_constants.h#L58

If you want to run this example again on the same board, you can modify the OTP rows used by the example - again, this will only work the first time for each set of OTP rows used

https://github.com/raspberrypi/pico-examples/blob/7e77a0c381863be0c49086567e7f1934d78ac591/otp/hello_otp/hello_otp.c#L18C1-L21C30

Octopus1633, 2024-09-26:
Thank you for your patience. After changing the address, I successfully ran the program again.
[image]

closed 2024-09-26 by Octopus1633

lurch, 2024-09-26:
@will-v-pi I wonder if it'd be useful if that example was modified to print out the existing OTP row contents _before_ atempting to write to them, and/or check that the rows are indeed empty before trying to modify them? :thinking:  (he blindly suggests, knowing very little about OTP!)

will-v-pi, 2024-09-26:
Yeah, that sounds sensible - if the rows are non-zero then it can just throw an error and stop