raspberrypi/pico-examples #562: Core #define statements mismatched

MadBerry58, 2024-10-10:
https://github.com/raspberrypi/pico-examples/blob/7e77a0c381863be0c49086567e7f1934d78ac591/freertos/hello_freertos/hello_freertos.c#L176

lurch, 2024-10-10:
Huh? Could you provide some more detail please? Thank you.

MadBerry58, 2024-10-11:
```
#if (configNUMBER_OF_CORES > 1)
    printf("Starting %s on both cores:\n", rtos_name);
    vLaunch();
#elif (RUN_FREE_RTOS_ON_CORE == 1 && configNUMBER_OF_CORES==1)
    printf("Starting %s on core 1:\n", rtos_name);
    multicore_launch_core1(vLaunch);
    while (true);
#else
    printf("Starting %s on core 0:\n", rtos_name);
    vLaunch();
#endif

    return 0;
```

Please correct me if I'm wrong, but in the above statements, the precompile directive `#if (configNUMBER_OF_CORES > 1)` will only generate the code 
```
    printf("Starting %s on both cores:\n", rtos_name);
    vLaunch();
    return 0;
```
thus not actually starting vLaunch on both cores.

The 2nd statement `#elif (RUN_FREE_RTOS_ON_CORE == 1 && configNUMBER_OF_CORES==1)` seems inconsistent, as `configNUMBER_OF_CORES==1` should not allow the core1(which sould be in theory dependent on a value of configNUMBER_OF_CORES==2) to be started, thus the generated code 
```
    printf("Starting %s on core 1:\n", rtos_name);
    multicore_launch_core1(vLaunch);
    while (true);
    return 0;
```
should also be logically inconsistent.

Feel free to correct any mistakes in my understanding, thank you