raspberrypi/pico-examples #572: Question about the example ir_nec

michcfr, 2024-11-16:
Hello
My question is about the example /pio/ir_nec
What parameters should I change if I want the carrier to be at 15Mhz instead of 38Khz?

thank you

lurch, 2024-11-16:
ping @mjcross who contributed this example in #129

mjcross, 2024-11-17:
I'll take a look

michcfr, 2024-11-17:
thank you. I would also change the transmission protocol.

mjcross, 2024-11-17:
@michcfr asked: 

> What parameters should I change if I want the carrier to be at 15Mhz instead of 38Khz?

If you want to use a different carrier frequency, the relevant parameters are in the calls to `nec_carrier_burst_program_init()` and `nec_carrier_control_program_init()` in [nec_transmit_library/nec_transmit.c](https://github.com/raspberrypi/pico-examples/blob/master/pio/ir_nec/nec_transmit_library/nec_transmit.c).

**However** be aware that you will need to find (or construct) an IR receiver module that supports your chosen carrier frequency. The device used in the example (VS1838b) only supports 38 kHz. You might find others at up to 56 kHz but I think 15 MHz is rather unlikely.

mjcross, 2024-11-17:
> I would also change the transmission protocol

That would certainly be outside the scope of this example.

mjcross, 2024-11-17:
@lurch close?

michcfr, 2024-11-17:
ok thank you very much

closed 2024-11-17 by michcfr