raspberrypi/pico-examples #573: Pico W tls_client tls_client_err -14

GoldeggCoder, 2024-11-17:
Recently, I attempted to transmit an HTTPS request using the Pico W. I cloned the example „tls_client“ and attempted to build it. This process was successful, but upon running the request on the Pico W, I encountered an error message indicating „tls_client_err -14“ after successfully resolving the domain. I am currently utilizing the Pico SDK version 2.0.

Potzi123, 2024-11-17:
I have the same Problem pls help

peterharperuk, 2024-11-18:
ERR_RST: the connection was reset by the remote host

GoldeggCoder, 2024-11-18:
But I am using the basic example with the domain provided.(worldtimeapi) Why does they close the connection?

peterharperuk, 2024-11-18:
It's not our website, but it appears to be down. We need to change it to something else.

GoldeggCoder, 2024-11-18:
Yeah thanks for the help. I got it working for my own domain. However it is very slow. It takes about 20 seconds to send a request. (See #411) Any fix for that?

lurch, 2024-11-25:
https://github.com/micropython/micropython/issues/16233 suggests that many sites refuse to respond to requests without a User-Agent header, so maybe https://github.com/raspberrypi/pico-examples/blob/master/pico_w/wifi/tls_client/picow_tls_client.c#L11 ought to include a `User-Agent` header too?

peterharperuk, 2025-01-15:
> Yeah thanks for the help. I got it working for my own domain. However it is very slow. It takes about 20 seconds to send a request. (See https://github.com/raspberrypi/pico-examples/issues/411) Any fix for that?

Make sure you're using a release build!

closed 2025-01-15 by peterharperuk