raspberrypi/pico-examples #575: Cmake executable issue w/Pico-examples Projects

benji7480, 2024-11-21:
I was able to compile all the pico-examples Projects 2 months ago. Now i tried doing the same and the experience build errors.
I have attached snapshots of the Errors when building below:

benji7480, 2024-11-21:
one of the problems here is that i am attaching the files/snapshots of the error, but, not sure if it is showing on the issue page.....

peterharperuk, 2024-11-21:
It's not. Can you not copy and paste the text?

benji7480, 2024-11-21:
> It's not. Can you not copy and paste the text?

it a snapshot/jpeg file Bro

benji7480, 2024-11-22:
> https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/attaching-files

Thanks Lurch for the hints, my files and all the limits are compliant, so my attached files shud be visible...One last thing i am using microsoft edge browser - is this browser supported or i shud use chrome,firfox,etc

benji7480, 2024-11-22:
i have resorted to text copy way as file attach doesnt work as peterharperuk suggested. Let the debug begin for my issue
hope some one can quikly solve it

closed 2024-11-25 by benji7480