raspberrypi/pico-examples #582: Cmake executable issue w/Pico-examples Projects

benji7480, 2024-11-25:
I was able to compile all the pico-examples Projects 2 months ago. Now i tried doing the same and the experience build errors.
I have copied text of the build Errors when building below:

build errors ---> Build failed: Unable to configure the project Bad CMake executable: "". Check to make sure it is installed or the value of the "cmake.cmakePath" setting contains the correct path

peterharperuk, 2024-11-25:
Can you try deleting your build folder and see if it's still a problem?

benji7480, 2024-11-25:
> Can you try deleting your build folder and see if it's still a problem?

Hi Peter, i am still not so profound/proficient w/the visual studio project folder structure.. tried my best to locate the build folder, but cudn't. can u give me a few simple steps to locate the build folder and then i can delete it.

here is also the additional copy of error on the o/p console : [main] Building folder: pico-examples 
[rollbar] Unhandled exception: Unhandled Promise rejection: build Error: Build failed: Unable to configure the project {}

benji7480, 2024-11-27:
> Can you try deleting your build folder and see if it's still a problem?

Hi Peter, located the build folder inside the pico-sdk folder abd deleted it. I gave the build cmd again and the same problem persists. pls guide further w/this issue and how to solve this..

peterharperuk, 2024-11-27:
Are you running on Windows or Linux or Mac?

benji7480, 2024-11-27:
> Hi Peter, i restored build folder inside the pico-sdk folder and deleted the build folder inside the pico-examples folder now. I gave the build cmd again and the same problem persists. 

i am running it on windows 11 home edition

peterharperuk, 2024-11-27:
@will-v-pi Have you heard of this error before?

will-v-pi, 2024-11-27:
No, but it sounds like the configuration is quite weird, especially if the build folder for pico-examples is inside the pico-sdk folder?

Could you check the value of `cmake.cmakePath` in your settings (the file is .vscode/settings.json) as suggested by the first error message?

benji7480, 2024-11-28:
> No, but it sounds like the configuration is quite weird, especially if the build folder for pico-examples is inside the pico-sdk folder?
> 
> Could you check the value of `cmake.cmakePath` in your settings (the file is .vscode/settings.json) as suggested by the first error message?

Hi Will-V-Pi, do not mis understand my communication chronology as "pico-examples folder is inside the pico-sdk folder". just that being a novice to VS environment i happened to find the SDK build folder 1st and deleted it. then i restored it and then found the pico-examples build folder and deleted it which is what Peter wanted me to do exactly. Will soon post the value of cmake.cmakePath in my .vscode/settings.json file....

benji7480, 2024-11-28:
> No, but it sounds like the configuration is quite weird, especially if the build folder for pico-examples is inside the pico-sdk folder?
> 
> Could you check the value of `cmake.cmakePath` in your settings (the file is .vscode/settings.json) as suggested by the first error message?

Hi Will-v-Pi, here is my .vscode/settings.json file contents below - i dont see any cmake.cmakePath line though...

`{
  // These settings tweaks to the cmake plugin will ensure
  // that you debug using cortex-debug instead of trying to launch
  // a Pico binary on the host
  "cmake.statusbar.advanced": {
    "debug": {
      "visibility": "hidden"
    },
    "launch": {
      "visibility": "hidden"
    },
    "build": {
      "visibility": "hidden"
    },
    "buildTarget": {
      "visibility": "hidden"
    }
  },
  "cmake.buildBeforeRun": true,
  "cmake.configureOnOpen": true,
  "cmake.configureSettings": {
    "CMAKE_MODULE_PATH": "${env:PICO_INSTALL_PATH}/pico-sdk-tools"
  },
  "cmake.generator": "Ninja",
  "C_Cpp.default.configurationProvider": "ms-vscode.cmake-tools"
}`

benji7480, 2024-11-28:
> You may have it configured more globally than that file then - could you see if you've modified it in your settings elsewhere? You can view settings from File -> Preferences -> Settings, then search for `cmake.cmakePath`.
> 
> If it's set to the default `cmake` then this suggests you don't have CMake installed in your Path - you can check this by running `cmake --version` from a terminal window.
> 
> Also, if you're using the deprecated pico-setup-windows installer (which I think you are given the comment at the top of that settings file), make sure you're using the Pico VS Code, rather than the regular VS Code.

Hi Will, found that it is set to default 'cmake' and i dont have cmake installed in my path.

 How do i now install Pico VS Code?

will-v-pi, 2024-11-28:
> How do i now install Pico VS Code?

That was for the deprecated pico-setup-windows installer, which would have added an alias called `Pico - Visual Studio Code` to the taskbar - if that's not installed then I would not recommend installing it now.

Instead of using that deprecated setup, I'd recommend using the new Pico VS Code Extension which is documented in the [Getting Started guide](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) PDF. Make sure you uninstall the old setup first (go to "Add or remove programs" and uninstall "Raspberry Pi Pico SDK v1.5.1" is it's there).

benji7480, 2024-11-28:
> That was for the deprecated pico-setup-windows installer, which would have added an alias called `Pico - Visual Studio Code` to the taskbar - if that's not installed then I would not recommend [...]

Hi Will,

tried this, and when i try to create the new project from examples by selecting blink, it throws up "failed to set up example"
this step shud get the new SDK installed for me and the environment, but it fails..Old sdk has been uninstalled prior

will-v-pi, 2024-12-02:
Yes, it looks like your system is still trying to use the old pico-setup-windows compiler. Make sure that is completely uninstalled before trying to use the new VS Code extension, including all files, registry entries, and environment variable settings.

benji7480, 2024-12-21:
> Yes, it looks like your system is still trying to use the old pico-setup-windows compiler. Make sure that is completely uninstalled before trying to use the new VS Code extension, including all files, registry entries, and environment variable settings.

Finally got it to work. thanks for all the hints will-v-pi

closed 2024-12-21 by benji7480