raspberrypi/pico-examples #584: Encrypted bootloader example does not work with stdio over USB

eliasbakken, 2024-11-29:
The bootloader/encrypted example works as expected when compiling it as is, data is transferred via uart. But when stdio is enabled for USB, there is only output when the enc_bootloader is run, not the final program. 

Adding `pico_enable_stdio_usb(hello_serial_enc 1)` to the CMakeLists.txt does not render any output via USB.

lurch, 2024-12-02:
ping @will-v-pi

will-v-pi, 2024-12-02:
This should be fixed when using picotool 2.1.0 (which is used by SDK 2.1.0) - see #558 and https://github.com/raspberrypi/picotool/pull/150

Could you try this with SDK & picotool 2.1.0 and see if you still have this issue?

eliasbakken, 2024-12-02:
Thank you for the suggestion. I did try it, but I do not see any change on the output of the USB.

eliasbakken, 2024-12-08:
It looks as though interrupts are not working when running from SRAM. 
The following program will not fire when loaded with the decoding bootloader, but will fire if run from flash directly. 
```
#include <stdio.h>
#include "pico/stdlib.h"

void gpio_callback(uint gpio, uint32_t events) {
    printf("GPIO %d\n", gpio);
}

int main() {
    stdio_uart_init();
    gpio_init(2);
    gpio_set_dir(2, GPIO_IN);
    gpio_set_irq_enabled_with_callback(2, GPIO_IRQ_EDGE_RISE | GPIO_IRQ_EDGE_FALL, true, &gpio_callback);

    while(true);
}
```

will-v-pi, 2024-12-17:
It looks like the issue is the `save_and_disable_interrupts` call in `rom_chain_image` in bootrom.h which seems to disable interrupts for the encrypted binary too.

This can be fixed by calling `restore_interrupts_from_disabled(0);` (and including `#include "hardware/sync.h"`) before calling `stdio_init_all()` in the encrypted binary.

```c
/**
 * Copyright (c) 2020 Raspberry Pi (Trading) Ltd.
 *
 * SPDX-License-Identifier: BSD-3-Clause
 */

#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/sync.h"

int main() {
    restore_interrupts_from_disabled(0);
    stdio_init_all();
    while (true) {
        printf("Hello, world!\n");
        printf("I'm an encrypted binary\n");
        sleep_ms(1000);
    }
}
```

@kilograham Is this a fix that should be added somewhere in the `runtime_init`, to ensure interrupts are always in a known state before a binary starts?

will-v-pi, 2024-12-18:
If you have UART enabled for the bootloader and USB enabled for the main program, then you can get output from the main program over USB.

If you have USB enabled for both, then you will only get USB output from the bootloader, as it's not currently possible to de-initialise then re-initialise the SDK stdio_usb, due to the lack of a `tusb_deinit` function in TinyUSB

eliasbakken, 2024-12-18:
Aha! Great, that works. Thank you.

closed 2025-05-30 by kilograham