raspberrypi/pico-examples #601: Restore Raspberry Pi Pico W Serial Drivers to Factory Settings

ThiagoSousa81, 2025-01-22:
I recently got a BitDogLab board, which has a Raspberry Pi Pico W as a microcontroller. I'm configuring it to run scripts in C. I saw an installation tutorial that teaches how to use Zadig to install the drivers for Interface 0 as USB Serial and Interface 1 as WinUSB. The problem is that when I put the USB Serial driver on interface 0, the serial only appears when the board is in bootsel and is not running the software. When the board is out of bootsel, Windows does not recognize it. I can't find the Serial working in any tested Serial client (PuTTY, VsCode...). I would like to know a way to restore these drivers to the factory settings if possible. I searched a lot on the internet, in the documentation and in forum posts, but I couldn't find anything about it. Some even told me that I shouldn't change the driver for Interface 0, but the tutorial said to do so.

As for other programs, I can easily control the GPIOs and upload .uf2 files, only Serial has this problem. If you can help me I would really appreciate it.

peterharperuk, 2025-01-22:
There are no drivers for Pico W. When you select bootsel it should show up as a disk drive in the Windows device manager. When running programs that are configured to support USB serial it will show up as a COM port / USB Serial Device. But the firmware has to be configured to support this - if it isn't then nothing will show up. This should all work without any additional drivers.

Can you post a link to the firmware (uf2 file) you are using and the instructions you're following?

lurch, 2025-01-22:
IMHO you should work your way through https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf and see how you get on with that. Note that there are many third-party tutorials online which haven't been updated to work with the latest versions of pico-sdk, so you should always treat any third-party documentation with a pinch of salt 😉