raspberrypi/pico-examples #625: FreeRTOS_Kernel_import.cmake uses old path

BrookdaleTech, 2025-03-14:
The freertos/FreeRTOS_Kernel_import.cmake is out of date. It does not have the current path for: FREERTOS_KERNEL_RP2040_RELATIVE_PATH. If I alter the code to change the path to match the current FreeRTOS- Kernel build it still throws an error: Directory '/home/james/Dev/FreeRTOS-Kernel' does not contain a 'rp2350-arm-s' port here: portable/GCC/ARM_CM33_NTZ.

BrookdaleTech, 2025-03-17:
After using your recommended file I see the following:
CMake Error at FreeRTOS_Kernel_import.cmake:77 (message):
  FreeRTOS location was not specified.  Please set FREERTOS_KERNEL_PATH.
Call Stack (most recent call first):
  CMakeLists.txt:38 (include)

Also, as requested I ran git submodule update --init

Understand where I am coming from -- I am out of my element and was following instructions, 2.3.7. FreeRTOS Ports, from Raspberry Pi Pico-series C/C++ SDK.

lurch, 2025-03-18:
> Understand where I am coming from -- I am out of my element and was following instructions, 2.3.7. FreeRTOS Ports, from Raspberry Pi Pico-series C/C++ SDK.

...which _does_ say "when building you should set `FREERTOS_KERNEL_PATH`" ?

I've just done a test on a fresh install of Raspberry Pi OS, and after doing:
```console
$ sudo apt install cmake gcc-arm-none-eabi libnewlib-arm-none-eabi build-essential
$ git clone https://github.com/raspberrypi/pico-sdk.git
$ git clone https://github.com/raspberrypi/pico-examples.git
$ git clone https://github.com/FreeRTOS/FreeRTOS-Kernel.git
$ export PICO_SDK_PATH=$HOME/pico-sdk
$ export FREERTOS_KERNEL_PATH=$HOME/FreeRTOS-Kernel
```
I was then able to do:
```console
$ cd pico-examples
$ mkdir build
$ cd build
$ cmake ..
$ cd freertos/hello_freertos
$ make
```
and the example FreeRTOS application built okay. I didn't get any errors, and it created `hello_freertos1.uf2` and `hello_freertos2.uf2`.

BrookdaleTech, 2025-03-18:
Okay it does build. Thanks