raspberrypi/pico-examples #639: mbedtls_sha256 example causes implicit function declaration warning

lurch, 2025-04-26:
E.g.
```
/opt/actions-runner/_work/pico-examples/pico-examples/pico-examples/sha/mbedtls_sha256/mbedtls_sha256.c: In function 'main':
/opt/actions-runner/_work/pico-examples/pico-examples/pico-examples/sha/mbedtls_sha256/mbedtls_sha256.c:35:14: warning: implicit declaration of function 'mbedtls_sha256_starts_ret'; did you mean 'mbedtls_sha256_starts'? [-Wimplicit-function-declaration]
   35 |     int rc = mbedtls_sha256_starts_ret(&ctx, 0);
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~
      |              mbedtls_sha256_starts
```

This seems to be due to https://github.com/Mbed-TLS/mbedtls/blob/development/docs/3.0-migration-guide.md#rename-mbedtls__ret-cryptography-functions-whose-deprecated-variants-have-been-removed (in combination with https://github.com/raspberrypi/pico-sdk/pull/2405 ).

I won't submit a PR myself, because I don't know what our stance is regarding support for backwards-compatibility with MbedTLS 2.x ?

peterharperuk, 2025-05-01:
> I don't know what our stance is regarding support for backwards-compatibility with MbedTLS 2.x

We allow the user to use their own library, so ideally we support old versions.

kilograham, 2025-05-29:
merged into develop

commit 6468b45f (2025-07-29): Use new sha function names in mbedtls example (#645)

Fixes #639