raspberrypi/pico-examples #640: Examples don't compile

TheRealDJ, 2025-04-29:
Well, blink doesn't at least ...
```
root@compiler:~/PICO/pico-examples/blink/build# echo $PICO_SDK_PATH
/root/PICO/pico-sdk
```
```
root@compiler:~/PICO/pico-examples/blink/build# ls $PICO_SDK_PATH
bazel        cmake           CONTRIBUTING.md  external  LICENSE.TXT   pico_sdk_init.cmake     README.md  test   WORKSPACE
BUILD.bazel  CMakeLists.txt  docs             lib       MODULE.bazel  pico_sdk_version.cmake  src        tools
```
```
root@compiler:~/PICO/pico-examples/blink/build# pwd
/root/PICO/pico-examples/blink/build
```
```
root@compiler:~/PICO/pico-examples/blink/build# cmake ..
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 13.3.0
-- The CXX compiler identification is GNU 14.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Using PICO_SDK_PATH from environment ('/root/PICO/pico-sdk')
PICO_SDK_PATH is /root/PICO/pico-sdk
Defaulting platform (PICO_PLATFORM) to 'rp2040' since not specified.
Defaulting target board (PICO_BOARD) to 'pico' since not specified.
Using board configuration from /root/PICO/pico-sdk/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
-- Defaulting build type to 'Release' since not specified.
Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m0plus_gcc' since not specified.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m0plus_gcc'
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
Build type is Release
-- Found Python3: /usr/bin/python3 (found version "3.12.3") found components: Interpreter 
TinyUSB available at /root/PICO/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /root/PICO/pico-sdk/lib/btstack
cyw43-driver available at /root/PICO/pico-sdk/lib/cyw43-driver
lwIP available at /root/PICO/pico-sdk/lib/lwip
mbedtls available at /root/PICO/pico-sdk/lib/mbedtls
CMake Warning at /root/PICO/pico-sdk/tools/Findpicotool.cmake:30 (message):
  No installed picotool with version 2.1.1 found - building from source

  It is recommended to build and install picotool separately, or to set
  PICOTOOL_FETCH_FROM_GIT_PATH to a common directory for all your SDK
[... 226 lines cut ...]
-- Configuring done (1.1s)
-- Generating done (0.1s)
-- Build files have been written to: /root/PICO/pico-examples/blink/build/_deps/picotool-build
[  9%] Performing build step for 'picotoolBuild'
[  2%] Built target errors
[ 85%] Built target mbedtls
[ 88%] Built target elf
[ 90%] Built target elf2uf2
[ 93%] Built target bintool
[100%] Built target picotool
[ 11%] Performing install step for 'picotoolBuild'
[  2%] Built target errors
[  4%] Built target elf
[ 88%] Built target mbedtls
[ 90%] Built target elf2uf2
[ 93%] Built target bintool
[100%] Built target picotool
Install the project...
-- Install configuration: "Release"
[ 12%] Performing test step for 'picotoolBuild'
picotool v2.1.2-develop (Linux, GNU-13.3.0, Release)
[ 13%] Completed 'picotoolBuild'
[ 13%] Built target picotoolBuild
make: *** [Makefile:91: all] Error 2
```

TheRealDJ, 2025-04-29:
To fix, I "edited" `CMakeLists.txt` ...
```
cmake_minimum_required(VERSION 3.13...3.27)

### change the below PROJECT variable to the name of your project
set(PROJECT blink)

set(PICO_SDK_PATH /root/PICO/pico-sdk)
set(PICO_BOARD pico2_w)

# initialize the SDK based on PICO_SDK_PATH
# note: this must happen before project()
include(pico_sdk_import.cmake)

project(${PROJECT})

# initialize the Raspberry Pi Pico SDK
pico_sdk_init()

# rest of your project

add_executable(
	${PROJECT}
	${PROJECT}.c
)

# Add pico_stdlib library which aggregates commonly used features
target_link_libraries(
	${PROJECT}
	pico_stdlib
	pico_cyw43_arch_none
)

# create map/bin/hex/uf2 file in addition to ELF.
pico_add_extra_outputs(${PROJECT})
```

lurch, 2025-04-29:
To make the CMake files easier to read, the examples in pico-examples aren't designed to be built individually, but only as part of the wider pico-examples repo. To create a standalone project, see Chapter 5 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf or the "Manually Create your own Project" section of Appendix C.

closed 2025-04-29 by lurch

TheRealDJ, 2025-04-29:
Thank you for clarifying. I have confirmed that I can compile all of the examples together. I don't see any building instructions in the readme ... Maybe something could be added to the readme?

DJ

lurch, 2025-04-29:
> Maybe something could be added to the readme?

The README already says "See [Getting Started with the Raspberry Pi Pico](https://rptl.io/pico-get-started) and the README in the [pico-sdk](https://github.com/raspberrypi/pico-sdk) for information on getting up and running."

If you have any suggestions for changes to the Getting Started guide, please file a new issue a https://github.com/raspberrypi/pico-feedback