raspberrypi/pico-examples #641: FreeRTOS - spp_streamer pi pico2 w

ispydre, 2025-04-30:
Hi all - 

I originally posted this issue on - https://github.com/bluekitchen/btstack/issues/682 - but this repo might be better suited for this issue. 

Background:
I'm trying to port spp_streamer.c to use FreeRTOS to my pi pico2 w. I made a project build with just the spp_streamer.c (bare-metal). I have tested that out by connecting my laptop (windows 11) to the pico2w over bluetooth. I open the serial monitor on VSCode, see the comport available, open that port and can get the test data.

Where I'm At:

Below is a core snippet of my FreeRTOS port.
```c
// FreeRTOS config

/* Scheduler Related */
#define configUSE_PREEMPTION                    1
#define configUSE_TICKLESS_IDLE                 0
#define configUSE_IDLE_HOOK                     0
#define configUSE_TICK_HOOK                     0
#define configTICK_RATE_HZ                      ( ( TickType_t ) 1000 )
#define configMAX_PRIORITIES                    32
#define configMINIMAL_STACK_SIZE                ( configSTACK_DEPTH_TYPE ) 512 
#define configUSE_16_BIT_TICKS                  0

#define configIDLE_SHOULD_YIELD                 1

/* Synchronization Related */
#define configUSE_MUTEXES                       1
#define configUSE_RECURSIVE_MUTEXES             1
#define configUSE_APPLICATION_TASK_TAG          0
#define configUSE_COUNTING_SEMAPHORES           1
#define configQUEUE_REGISTRY_SIZE               8
#define configUSE_QUEUE_SETS                    1
#define configUSE_TIME_SLICING                  1
#define configUSE_NEWLIB_REENTRANT              0
// todo need this for lwip FreeRTOS sys_arch to compile
#define configENABLE_BACKWARD_COMPATIBILITY     1
#define configNUM_THREAD_LOCAL_STORAGE_POINTERS 5

/* System */
#define configSTACK_DEPTH_TYPE                  uint32_t
#define configMESSAGE_BUFFER_LENGTH_TYPE        size_t

/* Memory allocation related definitions. */
#define configSUPPORT_STATIC_ALLOCATION         0
#define configSUPPORT_DYNAMIC_ALLOCATION        1
#define configTOTAL_HEAP_SIZE                   (128*1024)
#define configAPPLICATION_ALLOCATED_HEAP        0

/* Hook function related definitions. */
#define configCHECK_FOR_STACK_OVERFLOW          0
#define configUSE_MALLOC_FAILED_HOOK            0
#define configUSE_DAEMON_TASK_STARTUP_HOOK      0

/* Run time and task stats gathering related definitions. */
#define configGENERATE_RUN_TIME_STATS           0
#define configUSE_TRACE_FACILITY                1
#define configUSE_STATS_FORMATTING_FUNCTIONS    0

/* Co-routine related definitions. */
#define configUSE_CO_ROUTINES                   0
#define configMAX_CO_ROUTINE_PRIORITIES         1

/* Software timer related definitions. */
#define configUSE_TIMERS                        1
#define configTIMER_TASK_PRIORITY               ( configMAX_PRIORITIES - 1 )
#define configTIMER_QUEUE_LENGTH                10
#define configTIMER_TASK_STACK_DEPTH            1024

/* Interrupt nesting behaviour configuration. */
/*
#define configKERNEL_INTERRUPT_PRIORITY         [dependent of processor]
#define configMAX_SYSCALL_INTERRUPT_PRIORITY    [dependent on processor and application]
#define configMAX_API_CALL_INTERRUPT_PRIORITY   [dependent on processor and application]
*/

#if FREE_RTOS_KERNEL_SMP // set by the RP2xxx SMP port of FreeRTOS
/* SMP port only */
#ifndef configNUMBER_OF_CORES
#define configNUMBER_OF_CORES                   2
#endif
#define configNUM_CORES                         configNUMBER_OF_CORES
#define configTICK_CORE                         0
#define configRUN_MULTIPLE_PRIORITIES           1
#if configNUMBER_OF_CORES > 1
#define configUSE_CORE_AFFINITY                 1
#endif
#define configUSE_PASSIVE_IDLE_HOOK             0
#endif

/* RP2040 specific */
#define configSUPPORT_PICO_SYNC_INTEROP         1
#define configSUPPORT_PICO_TIME_INTEROP         1

#include <assert.h>
/* Define to trap errors during development. */
#define configASSERT(x)                         assert(x)

/* Set the following definitions to 1 to include the API function, or zero
to exclude the API function. */
#define INCLUDE_vTaskPrioritySet                1
#define INCLUDE_uxTaskPriorityGet               1
#define INCLUDE_vTaskDelete                     1
#define INCLUDE_vTaskSuspend                    1
#define INCLUDE_vTaskDelayUntil                 1
#define INCLUDE_vTaskDelay                      1
#define INCLUDE_xTaskGetSchedulerState          1
#define INCLUDE_xTaskGetCurrentTaskHandle       1
#define INCLUDE_uxTaskGetStackHighWaterMark     1
#define INCLUDE_xTaskGetIdleTaskHandle          1
#define INCLUDE_eTaskGetState                   1
#define INCLUDE_xTimerPendFunctionCall          1
#define INCLUDE_xTaskAbortDelay                 1
#define INCLUDE_xTaskGetHandle                  1
#define INCLUDE_xTaskResumeFromISR              1
#define INCLUDE_xQueueGetMutexHolder            1

#if PICO_RP2350
#define configENABLE_FPU                        1
#define configENABLE_MPU                        0
#define configENABLE_TRUSTZONE                  0
#define configRUN_FREERTOS_SECURE_ONLY          1
#define configMAX_SYSCALL_INTERRUPT_PRIORITY    16
#endif

// Bluetooth FreeRTOS Task
void bluetooth_task(void *pvParameters) {
    (void) pvParameters;
   
    int res = picow_bt_example_init();
    if (res){
        while (1) { tight_loop_contents(); }
    }

    // Call main BTstack setup
    picow_bt_example_main();
    btstack_run_loop_execute();

    // Should never reach here
    vTaskDelete(NULL);
}

void create_tasks(void) {
  xTaskCreate(bluetooth_task, "BluetoothTask", 4096, NULL, 2, &bluetooth_task_handle);

  vTaskStartScheduler();
}

int main( void )
{
    stdio_init_all();

    create_tasks();
}

// CMakeLists.txt  libraries
target_link_libraries(spp_freertos
  pico_stdlib
  pico_async_context_freertos
  pico_cyw43_arch_threadsafe_background
  pico_lwip_nosys
  pico_btstack_cyw43
  pico_btstack_classic
  FreeRTOS-Kernel
  FreeRTOS-Kernel-Heap4
)
``` 

I'm using spp_streamer.c, picow_bt_example_common.c. I'm able to connect to it - via my laptop and successfully view the test data 1/10 times. The other 9/10 times I'm able to connect to it - via my laptop but the com ports don't show up on the serial monitor.

I believe there might be some setup calls i might need to change to fully get freertos working with spp_streamer. Happy to read more documentation, additional examples, or try any suggestions here.

Thanks!

peterharperuk, 2025-04-30:
Presumably you have modified pico-examples directly to try this? Can you attach a patch so I don't have to copy and paste your changes?

In theory you can build the examples for freertos by using the following cmake command line: cmake .. -DBTSTACK_EXAMPLE_TYPE=freertos

However you will need this fix https://github.com/raspberrypi/pico-examples/pull/643

mringwal, 2025-04-30:
Isn't FreeRTOS one of the options for the run mode in PicoSDK?
(I remember: polling, background task, FreeRTOS, while I usually went along with polling)

peterharperuk, 2025-04-30:
Yes. It worked for me first time.

ispydre, 2025-04-30:
Currently not at my machine till later today. But I overlooked the freertos example on the pi pico 2 w. I thought didn’t see it called the btstack_main. I would have tested this out first before making my own project. 

I’ll test the:

cmake .. -DPICO_BOARD=pico2_w -DBTSTACK_EXAMPLE_TYPE=freertos

With the spp_streamer example later today and report back. 

Appreciate the help!

ispydre, 2025-05-01:
@peterharperuk Ran the example with the fix you suggested. Works great. Thanks for the help

closed 2025-05-01 by ispydre