raspberrypi/pico-examples #642: Official flash nuke doesn't work, but https://github.com/Gadgetoid/pico-universal-flash-nuke does work

tyeth, 2025-04-30:
Got an Adafruit RP2040 Adalogger Feather, and it previously had the Adafruit Wippersnapper offline firmware on it, which uses/creates a small FAT user-data partition to share with the OS. 
I used the official flash nuke, then installed a new version of the firmware, but the file system partition showed up on the windows host with the old files still present (config-old.json etc). I repeated the flash nuke downloading it from the pi site instead of adafruit learn guide copy, with the same result.
Then knowing about Gadgetoid's alternative I used that (from https://github.com/Gadgetoid/pico-universal-flash-nuke ), and then installed the new firmware and it then showed the user file system had been correctly erased.

Why does the official one fail to nuke the board entirely?

tyeth, 2025-04-30:
Recreation steps:

1. Download official flash nuke from https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#resetting-flash-memory
2. Boot board in bootloader mode, flash nuke, then Install an appropriate rp2 firmware from https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/releases/tag/1.0.0-offline-beta.1
3. Wait for WIPPER drive to show up, then create some meaningless text file called hello.txt
4. reboot device and see it reappear with file.
5. reset device into bootloader mode and flash nuke it, then install firmware again.
6. see WIPPER device appear with hello.txt present.
7. Repeat with Steps 2-6 with Gadgetoids flashnuke instead: https://github.com/Gadgetoid/pico-universal-flash-nuke/releases/download/v1.0.1/universal_flash_nuke.uf2

peterharperuk, 2025-04-30:
Our version uses PICO_FLASH_SIZE_BYTES which assumes the flash size is 16Mbits

tyeth, 2025-04-30:
That makes sense thanks.
It's not made clear anywhere in the documentation, so maybe could be, but better than that maybe the flash nuke UF2 / example can be improved to work for all flash sizes via some form of auto-detection. 
I'm curious what would happen if the size was assumed larger than reality, would it just fail and then watchdog reboot (i.e. effectively succeed in flash nuking).

peterharperuk, 2025-04-30:
You're probably right. 

> I'm curious what would happen if the size was assumed larger than reality, would it just fail and then watchdog reboot (i.e. effectively succeed in flash nuking assuming the flash_nuke app didn't load next boot).

I think it might just wrap around and probably work.

lurch, 2025-04-30:
> Our version uses PICO_FLASH_SIZE_BYTES which assumes the flash size is 16Mbits

@will-v-pi Does this mean that the universal flash nuke will wipe the whole flash on a Pico 1, but only the first half of flash on a Pico 2?

peterharperuk, 2025-04-30:
PICO_FLASH_SIZE_BYTES is correct for pico2

will-v-pi, 2025-04-30:
I think the issue here is that `PICO_FLASH_SIZE_BYTES` (used by flash_nuke) is configured by the board header, so is only 1M on Pico and 2M on Pico 2. The universal flash_nuke didn't change that define, so it is only erasing 1M of flash on RP2040s and 2M on RP2350s.

I think the fix would be to add a new example that just erases all 16M of flash (and maybe the CS1 window too on RP2350, if it's configured), and then build nuke_universal from that example instead

Gadgetoid, 2025-05-01:
> I think the fix would be to add a new example that just erases all 16M of flash (and maybe the CS1 window too on RP2350, if it's configured), and then build nuke_universal from that example instead

I should probably have made an upstream PR for:

https://github.com/Gadgetoid/pico-universal-flash-nuke/blob/0fb758b9c46c2c98b52faeb10d5168b169a820f6/nuke.c#L29-L37

Not exactly rocket surgery, but I was floundering about desperate for relevance and wanted to Make A Thing.

tyeth, 2025-06-13:
@peterharperuk any chance of merging #653 and then releasing an updated UF2 onto the main raspberrypi.com site?