raspberrypi/pico-examples #648: freertos example fails to build on pico2 when core number set to 2.

ObligedTester, 2025-05-14:
Hello,

Having issues with building the freeRTOS example on pico2.

It builds fine with core count set to 1, and it also builds fine on pico 1 both single and dual core

Something seems to be up with the "portmacro". Its like in the cases where core is set to 2 in rp2035 it causes this to throw alot of errors.

Output:

[image]

peterharperuk, 2025-05-14:
Please clarify what you are setting to 2. configNUMBER_OF_CORES is currently set to 2 and it builds fine. It looks like you have something wrong in your build as pic.h belongs to another processor.

ObligedTester, 2025-05-14:
As per the example, it is set to "=1" in the cmakelists file in this repository. Did you change this?

It does not matter if omit this line and set it directly in the freertos config file or not. If compiled for the pico2 it fails for dual core. Works for single core. And weirdly enough works for pico1 multicore.

The pic.h reference is strange, might very well be my pull that is incorrect. Will attempt to pull everything once more.

But it's still weird that it fails only for multi core pico2 and nothing else.

kilograham, 2025-05-14:
How about including what commit of FreeRTOS you are using from where; what your build configuration is etc.

ObligedTester, 2025-05-14:
So the kernel that is used is the 11.2.0 (Mar 4 release) and its from here: https://github.com/FreeRTOS/FreeRTOS-Kernel
Likewise, the RP2350 port is from there: https://github.com/FreeRTOS/FreeRTOS-Kernel-Community-Supported-Ports/tree/bae4c7aa19009825ba48071a8fe25dcb8be84880/GCC

My CMakeLists for the project look like this:
```
# == DO NOT EDIT THE FOLLOWING LINES for the Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.1.1)
set(toolchainVersion 14_2_Rel1)
set(picotoolVersion 2.1.1)
set(picoVscode ${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
if (EXISTS ${picoVscode})
    include(${picoVscode})
endif()
# ====================================================================================
set(PICO_BOARD pico2 CACHE STRING "Board type")

cmake_minimum_required(VERSION 3.13)

# Pull in SDK (must be before project)
include(pico_sdk_import.cmake)

# Pull in FreeRTOS
set(FREERTOS_KERNEL_PATH ${CMAKE_CURRENT_LIST_DIR}/kernel)
include(FreeRTOS_Kernel_import.cmake)

# Project name
project(pico-freertos)

pico_sdk_init()

# Configuration projects
add_executable(${PROJECT_NAME}
    main.c
)

target_include_directories(${PROJECT_NAME} PRIVATE
kernel_include/

)

target_compile_definitions(${PROJECT_NAME} PRIVATE configNUMBER_OF_CORES=2 )

target_link_libraries(${PROJECT_NAME} PRIVATE
    pico_stdlib
    pico_multicore
    FreeRTOS-Kernel-Heap4
)

target_compile_options(${PROJECT_NAME} PRIVATE -Wall)

# Run the entire project in SRAM
# pico_set_binary_type(${PROJECT_NAME} copy_to_ram)

pico_add_extra_outputs(${PROJECT_NAME})
```

My freertosconfig.h is copy of the one from the examples and likewise the freertos_kernel_import.cmake.

I might have forgotten something to include here, if so please remind me. Been a while since i worked with pico/freeRTOS.

lurch, 2025-05-15:
Hmmm, looks like @ObligedTester is using a `CMakeLists.txt` that has been generated by https://github.com/raspberrypi/pico-vscode (the `DO NOT EDIT THE FOLLOWING LINES` is  bit of a [giveaway](https://github.com/raspberrypi/pico-vscode/blob/main/scripts/pico_project.py#L43) ) which IIRC is designed to build each example as a standalone project; whereas @peterharperuk is presumably using the "CLI build" of pico-examples, and so will be using [this CMakeLists.txt](https://github.com/raspberrypi/pico-examples/blob/develop/freertos/hello_freertos/CMakeLists.txt) (which does indeed [set `configNUMBER_OF_CORES=1`](https://github.com/raspberrypi/pico-examples/blob/develop/freertos/hello_freertos/CMakeLists.txt#L20)) which IIRC is designed to build each example only as part of the wider pico-examples project.

(I'm afraid I don't have much experience of either the Pico VS Code Extension nor FreeRTOS, so can't offer any further advice at this point)

ObligedTester, 2025-05-15:
lurch, 
Thanks for your input!

Yep, that is correct, mine is standalone. I imported mine with the beautiful vscode extension, so it added the top part of the cmakelists file automatically. When I saw everything worked besides RP2350 dual core, and that the example defaulted to single core, I got a suspicion that maybe something is up with the example.

peterharperuk, 2025-05-15:
Please post all your code in a zip or a link to a github repo. You're wasting our time guessing what your problem might be. You're wrong about the example - it builds two versions. One which uses 1 core and another that uses 2 cores.

lurch, 2025-05-15:
> You're wrong about the example - it builds two versions. One which uses 1 core and another that uses 2 cores.

Ooops, _mea culpa_! 🤦‍♂  I hadn't spotted that https://github.com/raspberrypi/pico-examples/blob/develop/freertos/hello_freertos/CMakeLists.txt actually defines both a [hello_freertos1](https://github.com/raspberrypi/pico-examples/blob/develop/freertos/hello_freertos/CMakeLists.txt#L1) target which explicitly sets [`configNUMBER_OF_CORES=1`](https://github.com/raspberrypi/pico-examples/blob/develop/freertos/hello_freertos/CMakeLists.txt#L20) and **also** a [hello_freertos2](https://github.com/raspberrypi/pico-examples/blob/develop/freertos/hello_freertos/CMakeLists.txt#L24) target which _doesn't_ set `configNUMBER_OF_CORES` (which means that it'll use the default [`#define configNUMBER_OF_CORES 2`](https://github.com/raspberrypi/pico-examples/blob/develop/freertos/FreeRTOSConfig_examples_common.h#L107-L109) ).

ObligedTester, 2025-05-15:
Sure, here comes the zip: 

[freertos2350_multicoreissue.zip](https://github.com/user-attachments/files/20227467/freertos2350_multicoreissue.zip)

peterharperuk, 2025-05-15:
There's something wrong with your "kernel" folder. Can you try doing the following?

```
mv kernel kernel_broken
git clone https://github.com/FreeRTOS/FreeRTOS-Kernel.git -b main --recurse-submodules kernel

```

It then builds fine for me.

ObligedTester, 2025-05-15:
Alright! That did the trick.
I have no clue what might have happened with the repositories. I know the freertos-kernel itself has been updated via git a couple of times since i "first" cloned it. But dont think that would have anything to do with it.
Very weird issue!

Thanks a lot everyone involved!

closed 2025-05-15 by ObligedTester

lurch, 2025-05-16:
> I have no clue what might have happened with the repositories.

It looks like your `kernel` submodule was simply out of date? (which is easily done, as git submodules can be quite confusing when you first encounter them). If I follow Peter's instructions above, then the start of `kernel_broken/History.txt` (i.e. the one that was in your zipfile) says:
```
Changes between FreeRTOS V11.0.1 and FreeRTOS V11.1.0 released April 22, 2024
```
but the start of `kernel/History.txt` (i.e. the latest one from the FreeRTOS-Kernel repo) says:
```
Changes between FreeRTOS V11.1.0 and FreeRTOS V11.2.0 released March 04, 2025
```
i.e. it looks like you were using FreeRTOS V11.1.0 when you **should** have been using FreeRTOS V11.2.0 ?
The changes listed in `kernel/History.txt` for V11.2.0 include "Update CMake files for RP2040 port to fetch the port from the Community-Supported-Ports repo." and "Update the RP2040 port to add support for Raspberry Pi Pico SDK 2.0.0.", and Pico SDK 2.0.0 was the first version to support RP2350 / Pico2.

lurch, 2025-05-16:
> as git submodules can be quite confusing when you first encounter them

@ObligedTester You might find the sample `CMakeLists.txt` here to be helpful? https://github.com/raspberrypi/pico-sdk/issues/1659#issuecomment-2886559548

ObligedTester, 2025-05-17:
Hey!

Thanks for your further help and input!

Yep something must have just gone wrong when I cloned the kernel repos.. not easy to know exactly why, at least it's fixed now.

I have changed my setup somewhat now. As I have several freertos projects currently, I'm just cloning a "main" one, for all different projects to use. Like it's intended to be done :D