raspberrypi/pico-examples #657: Using custom pins in PIO HUB75 example

Fantonrko, 2025-06-23:
Hi, is it possible to use custom pins for PIO HUB75 ?
I am currently using only R0, G0, B0 in the code, so I could change some parts of the code and remove R1, G1, B1, but is it possible to use custom pins, like IO5, IO2, IO3 (as RGB pins)?

Fantonrko, 2025-07-13:
Sorry for the long reply, Unfortunately I couldn't. I saw that the PIO program uses serial pins, but I don't quite understand how I can get rid of R1, G1, B1 pins, since my matrix does not use them (my matrix contains only 9 rows)

lurch, 2025-07-14:
I don't have one of these boards myself (and I see that `hub75.c` hasn't been modified since January 2021!), so this is just a wild guess from a quick look at the source code...
Looking at https://github.com/raspberrypi/pico-examples/blob/evelop/pio/hub75/hub75.c I see that it does:
```
#define DATA_BASE_PIN 0
#define DATA_N_PINS 6
```
which seems to correspond to R0, G0, B0, R1, G1, B1. It then passes `DATA_BASE_PIN` to `hub75_data_rgb888_program_init` but it doesn't make use of `DATA_N_PINS` at all. So _maybe_ you need to add an extra `rgb_n_pins` parameter to the `hub75_data_rgb888_program_init` function in https://github.com/raspberrypi/pico-examples/blob/develop/pio/hub75/hub75.pio and change all the places in that function where it says `6` to instead use `rgb_n_pins`. And then you could change `DATA_N_PINS` to 3.
But I've **no** idea if that'll work or not, or if it'll end up breaking some other part of the code. If it doesn't work, you'll probably be better off asking for help at https://forums.raspberrypi.com/viewforum.php?f=143 rather than on GitHub.