raspberrypi/pico-examples #663: Importing picow_ble_temp_sensor and picow_ble_temp_reader are imported wrong by vscode

peterharperuk, 2025-07-07:
The BT example in pico_w/bt/standalone has two examples picow_ble_temp_sensor and picow_ble_temp_reader.

If I import and debug picow_ble_temp_sensor it starts in client.c which belongs to picow_ble_temp_reader

picow_ble_temp_reader seems to use server.c which belongs to picow_ble_temp_sensor  so it's as if these two are imported the wrong way around?

lurch, 2025-07-07:
Looks like https://github.com/raspberrypi/pico-vscode/blob/main/data/0.17.0/examples.json#L1433-L1454 might have something to do with it, but that provides no clues as to why it gets things the wrong way around!

will-v-pi, 2025-07-08:
The import is done by the [pico_project.py](https://github.com/raspberrypi/pico-vscode/blob/main/scripts/pico_project.py) script - it looks like the issue here is because there's a WiFi example in the same CMakeLists.txt file, it's triggering project name replacement mechanics which only work if the file only contains a single pair of `xxx_threadsafe_background` and `xxx_poll` executables.

The simplest fix is to to enable the CMake Tools Extension integration (can be done by clicking Import Project, and then re-importing the example and ticking Enable CMake Tools Extension integration) - this then lets you select the example you want to run & debug

closed 2025-07-15 by will-v-pi