raspberrypi/pico-examples #670: RP2350 Hello Universal ARM + RISC-V 2.1.1

5HT, 2025-07-16:
Greetings!

Help me build ARM/RISC-V Universal Binary using official master branches 2.1.1/sdk-2.1.1

5HT, 2025-07-16:
I actually have twp binaries:

```
tonpa@TRISTELLAR:~/depot/bitedits/pico-examples$ picotool info -a build-arm/hello_world/usb/hello_usb.uf2
File build-arm/hello_world/usb/hello_usb.uf2 family ID 'rp2350-arm-s':

Program Information
 name:                hello_usb
 web site:            https://github.com/raspberrypi/pico-examples/tree/HEAD/hello_world/usb
 features:            USB stdin / stdout
 binary start:        0x10000000
 binary end:          0x100054d0
 target chip:         RP2350
 image type:          ARM Secure

Fixed Pin Information
 none

Build Information
 sdk version:         2.1.1
 pico_board:          pico2
 boot2_name:          boot2_w25q080
 build date:          Jul 16 2025
 build attributes:    Release

Metadata Block 1
 address:             0x10000138
 next block address:  0x100054bc
 block type:          image def
 target chip:         RP2350
 image type:          ARM Secure

Metadata Block 2
 address:             0x100054bc
 next block address:  0x10000138
 block type:          ignored
tonpa@TRISTELLAR:~/depot/bitedits/pico-examples$ picotool info -a build-riscv/hello_world/usb/hello_usb.uf2
File build-riscv/hello_world/usb/hello_usb.uf2 family ID 'rp2350-riscv':

Program Information
 name:                hello_usb
 web site:            https://github.com/raspberrypi/pico-examples/tree/HEAD/hello_world/usb
 features:            USB stdin / stdout
 binary start:        0x10000000
 binary end:          0x10006440
 target chip:         RP2350
 image type:          RISC-V

Fixed Pin Information
 none

Build Information
 sdk version:         2.1.1
 pico_board:          pico2
 boot2_name:          boot2_w25q080
 build date:          Jul 16 2025
 build attributes:    Release

Metadata Block 1
 address:             0x10000014
 next block address:  0x1000642c
 block type:          image def
 target chip:         RP2350
 image type:          RISC-V
 entry point:         EP 0x10000036, SP 0x20082000

Metadata Block 2
 address:             0x1000642c
 next block address:  0x10000014
 block type:          ignored
```

How can I merge it?

5HT, 2025-07-16:
I found reference image: https://datasheets.raspberrypi.com/soft/hello_world.uf2

```
tonpa@TRISTELLAR:~/depot/bitedits/be/arch/riscv$ picotool info -a hello_world.uf2
File hello_world.uf2 family ID 'rp2040':

Program Information
 name:                  hello_universal
 features:              UART stdin / stdout
                        USB stdin / stdout
 binary start:          0x10000000
 binary end:            0x10005dd0
 target chip:           RP2350
 image type:            ARM Secure

Fixed Pin Information
 0:                     UART0 TX
 1:                     UART0 RX

Build Information
 sdk version:           2.0.0
 pico_board:            pico
 build date:            Aug 15 2024
 build attributes:      Release

Metadata Block 1
 address:               0x100001e8
 next block address:    0x10005dd0
 block type:            ignored

Metadata Block 2
 address:               0x10005dd0
 next block address:    0x10006138
 block type:            ignored

Metadata Block 3
 address:               0x10006138
 next block address:    0x1000d650
 block type:            image def
 target chip:           RP2350
 image type:            ARM Secure

Metadata Block 4
 address:               0x1000d650
 next block address:    0x1000e014
 block type:            image def
 target chip:           RP2350
 image type:            ARM Secure
 rolling window delta:  0x00006000
 vector table:          0x10000000

Metadata Block 5
 address:               0x1000e014
 next block address:    0x10019108
 block type:            image def
 target chip:           RP2350
 image type:            RISC-V
 entry point:           EP 0x10000036, SP 0x20082000

Metadata Block 6
 address:               0x10019108
 next block address:    0x100001e8
 block type:            image def
 target chip:           RP2350
 image type:            RISC-V
 rolling window delta:  0x0000e000
 entry point:           EP 0x10000036, SP 0x20082000

-------------------------------------------
File hello_world.uf2 family ID 'absolute':

Program Information
 name:                  hello_universal
 features:              UART stdin / stdout
                        USB stdin / stdout
 binary start:          0x10000000
 binary end:            0x10005dd0
 target chip:           RP2350
 image type:            ARM Secure

Fixed Pin Information
 0:                     UART0 TX
 1:                     UART0 RX

Build Information
 sdk version:           2.0.0
 pico_board:            pico
 build date:            Aug 15 2024
 build attributes:      Release

Metadata Block 1
 address:               0x100001e8
 next block address:    0x10005dd0
 block type:            ignored

Metadata Block 2
 address:               0x10005dd0
 next block address:    0x10006138
 block type:            ignored

Metadata Block 3
 address:               0x10006138
 next block address:    0x1000d650
 block type:            image def
 target chip:           RP2350
 image type:            ARM Secure

Metadata Block 4
 address:               0x1000d650
 next block address:    0x1000e014
 block type:            image def
 target chip:           RP2350
 image type:            ARM Secure
 rolling window delta:  0x00006000
 vector table:          0x10000000

Metadata Block 5
 address:               0x1000e014
 next block address:    0x10019108
 block type:            image def
 target chip:           RP2350
 image type:            RISC-V
 entry point:           EP 0x10000036, SP 0x20082000

Metadata Block 6
 address:               0x10019108
 next block address:    0x100001e8
 block type:            image def
 target chip:           RP2350
 image type:            RISC-V
 rolling window delta:  0x0000e000
 entry point:           EP 0x10000036, SP 0x20082000
```

5HT, 2025-07-16:
Can you fix me or master?

5HT, 2025-07-17:
I can also confirm that default build with built images from https://github.com/raspberrypi/pico-sdk-tools/releases/download/v2.1.1-2/riscv-toolchain-15-aarch64-lin.tar.gz works in full automatic mode OK in `pico-examples`

```
cmake -DPICO_BOARD=pico2 -DPICO_PLATFORM=rp2350-riscv \
      -DPICO_RISCV_TOOLCHAIN_PATH=/home/maxim/depot/raspberrypi/riscv32-unknown-elf \
      -DPICO_ARM_TOOLCHAIN_PATH=/usr \
      -DCMAKE_C_COMPILER=/home/maxim/depot/raspberrypi/riscv32-unknown-elf/bin/riscv32-unknown-elf-gcc \
      -DCMAKE_CXX_COMPILER=/home/maxim/depot/raspberrypi/riscv32-unknown-elf/bin/riscv32-unknown-elf-g++ ..
```

lurch, 2025-07-17:
@5HT Does it work any better if you use the `develop` branches of `pico-sdk`, `pico-examples` and `picotool` ?

5HT, 2025-07-17:
I will try, but I thought `2.1.1` and `master` need more attention.

lurch, 2025-07-17:
> I will try, but I thought `2.1.1` and `master` need more attention.

`master` and `2.1.1` are (currently) the same thing. We're currently working on `develop`, which will "soon" become `2.1.2` and the new `master`.

will-v-pi, 2025-07-17:
I've just tested with `develop` branches of `pico-sdk`, `pico-examples` and `picotool` and it is working fine on Debian Bookworm. Also worked fine for me on the `master` branches. I can confirm there is an issue with linking only 2 binaries due to bugs in picotool's command line handling.

@5HT Can you confirm that you definitely deleted the whole `build` directory in `pico-examples` before compiling - the universal examples use CMake External Projects, so you often need to delete the whole `build` directory if you change anything

5HT, 2025-07-17:
Compilation on Debian Bookworm works fine, but UF2 image can't be flashed:
This also included in report

[image]

will-v-pi, 2025-07-17:
It's possible your Risc-V toolchain is broken, given the error is always occurring when compiling the Risc-V binary. You could try using the latest toolchain from [pico-sdk-tools](https://github.com/raspberrypi/pico-sdk-tools/releases) instead, which is what I'm using.

5HT, 2025-07-17:
No, That is not! I'm trying to flash UF2 as stated in report. I can attach it.

5HT, 2025-07-17:
Can you read messages before writing?

5HT, 2025-07-17:
I don't have time for this. I will figure out sooner ore later if you will be able to fix this.

5HT, 2025-07-17:
Bye.

will-v-pi, 2025-07-17:
No, in your screenshot it clearly shows "Copying file: hello_universal.bin" - that is not a UF2 file

Closing now, as you don't have time, and we can't reproduce the issue

closed 2025-07-17 by will-v-pi